FilipDominec/rp2daq #1: please consider current source DAC LTC2662 (ADI)

caizikun, 2021-08-19:
please consider current source DAC LTC2662 (ADI) with fast sweep.
https://www.analog.com/en/products/ltc2662.html
Thank you!

FilipDominec, 2021-08-30:
The manufacturer seems to be literally "ahead of what is possible". Although it is officially in production, I could not find LTC2662 on Farnell, Mouser, TME nor whole Ebay. My policy is to keep all extensions tested experimentally, so I cannot help you as of yet.

Thanks for the tip, anyway. Its parameters are nice and price is also fine.

FilipDominec, 2023-10-03:
Dear @caizikun, if I implement rudimentary SPI messaging, would you still be interested in testing it on a LTC2662? 

Very cheap chips I buy without hesitation just for testing; the latter is unfortunately not one of them. Currently I am not going to buy it, so I would have to rely on collaboration with you.

caizikun, 2023-10-10:
@FilipDominec  Sorry, I just leave my last job and have no suitable debugging environment. But as i know, multi-channels current source would be more popular for DAQ card application. Maybe you can try to prepare the register setting API. If I have debugging environment, i will tell you.

FilipDominec, 2023-10-10:
OK, thanks.  

Meanwhile I decided to demonstrate DAC capabilities with [TDA1543](https://www.alldatasheet.com/datasheet-pdf/pdf/19214/PHILIPS/TDA1543.html). Though far from being high-end, it receives data over I2S present in Pi Pico; it has 2×16 bit outputs, comes in convenient DIP package and costs near to nothing. Of course other I2S devices should be compatible as well.

Further DAC or other peripheral tips are welcome.