FilipDominec/rp2daq #4: Suggestion

epsi1on, 2023-07-31:
First of all thanks and congrats for the innovative project.
Currently the compiled firmware (I mean `/build/rp2daq.uf2` file) is placed inside the project files. I would like to suggest you to publish the compiled binary file as a release file. I think this is how usually the binary files are published in github.
also the version control would be easier and files can be visible from release section:
https://github.com/FilipDominec/rp2daq/releases

FilipDominec, 2023-08-01:
Thanks, I will do it like yuo describe. Currently, the automatic push of new binary firmware on every ABI change is inefficient.

closed 2023-08-01 by epsi1on

FilipDominec, 2023-10-07:
Done. For the latest current version of the pre-compiled firmware, please always refer to the link:

> https://github.com/FilipDominec/rp2daq/releases/latest/download/rp2daq.uf2

I also updated README.md to reflect this. 

Thanks for your suggestion.

epsi1on, 2023-10-07:
Wonderful.
I should put a check for firmaware version in scope. Does the returned firmaware version from PICO do change for each individual release?

FilipDominec, 2023-10-08:
It should represent the latest date the firmware ABI was updated (breaking binary compatibility), but haven't been overly conscientious about it in previous months.

Anyway, the latest release as well as current git version of rp2daq return human-readable 6-byte string "231005".

epsi1on, 2023-10-08:
ok, thanks

epsi1on, 2025-01-22:
Dear Flip,
Could you please put the updated binary uf2 file of firmware into release section of repo.
Current binary ([v0.2.0 easier GPIO control](https://github.com/FilipDominec/rp2daq/releases/tag/v0.2.0) [L](https://github.com/FilipDominec/rp2daq/releases/latest)) relates to the time before the low impedance problem solved in #18.
Actually i have up-to-date version locally, but i think it would be nice to have it on release section too.
I rather to do not include the binary firmware into my oscilloscope app, but rather to put hyper link for download.

Thanks

FilipDominec, 2025-01-22:
Dear Ehsan, I shall make a new release within a week.

FilipDominec, 2025-03-07:
Sorry. I am fixing some last bugs, the 0.3.0 issue will come soon!