FilipDominec/rp2daq #11: STM32

epsi1on, 2023-10-11:
Hello there,
Do you know any firmware that utilities use stm32's ADC for using as sampler for osciloscope?

Thanks

FilipDominec, 2023-10-11:
Hi, I have no knowledge of such firmware, but in its rudimentary form it should not be hard to write if you are familiar with C. I am not sure it is a good investment of your time, though:

* My experience with rp2daq is that it is far from trivial to ensure continuous 500 ksps data flow from Pi Pico, when Windows & Linux behave a bit differently under 100% CPU load. I have accomplished this, and I have no capacity to extend my projects to STM32. 

* What I definitely intend to implement is 100 Msps oscilloscope using [AD9288-100](https://www.ebay.com/sch/i.html?_nkw=AD9288BSTZ-100) and the Pi Pico's PIO&DMA features. Of course this data rate will no more fit into the 1 MB/s bandwidth of Pico's slow USB, it will have to operate either in triggered bursts, or use some data binning and "digital phosphor screen". This will make it nearly competitive with the low-end commercial oscilloscopes.

FilipDominec, 2023-10-11:
With some [radical overclocking of Pico](https://www.raspberrypi.com/news/dont-try-this-at-home-overclocking-rp2040-to-1ghz/), and 74€ investmesnt in [AD9286BCPZ-500](https://eu.mouser.com/ProductDetail/Analog-Devices/AD9286BCPZ-500?qs=%2FtpEQrCGXCyIYIpUGZonBw%3D%3D), one could go up to some 500 Msps; the PIO takes parallel 8bit data on the full system clock. But the reliability of such setup would have to be established experimentally.

FilipDominec, 2023-10-11:
Perhaps a more realistic option is to stick to 250 Msps, corresponding to the current reliable system clock of 250 MHz. 

Such ADC seem available for some 20-25€, like [AD9481BSUZ-250](https://www.ebay.com/sch/i.html?_nkw=AD9481%20ADC). Of course these may be Chinese counterfeits, but for a hobby-grade oscilloscope and concept testing they may serve well. 

So far I have bought [AD9288-100](https://www.ebay.com/sch/i.html?_nkw=AD9288BSTZ-100) and will test it as I get some spare time again.

closed 2023-10-11 by FilipDominec

epsi1on, 2023-10-11:
Cool, I believe with something like 100Msps, your project can be more popular.
You can make another opensource project for PCB hardware with like AD9288. 
Then I think you can also put the assembled PCB for sale with small profit margin, as there will be potentially many users which are willing to buy pcb and make a 100Msps DAQ with RPI Pico. First one would be me if price is less than or equal $50 :)

FilipDominec, 2023-10-11:
Yes, I believe such a $15 scope could get quite popular. 

For some uses, e.g. in digital electronics, it would be even more appropriate than common scopes. Imagine how many experiments you could run with additional hardware interfaces of the RP2! 

A dedicated PCB would be cool, but it is not necessary - one can just master some soldering (use a lot of flux!) and get a TQFP breakout board for $3.

epsi1on, 2024-01-17:
> Yes, I believe such a $15 scope could get quite popular.

yeah, i can say it would lift the market. But about lifting the market, I can tell you it would not be easy. You must pass over almost all current limits. You definitely need to strongly concentrate on it. Good luck anyways...