FilipDominec/rp2daq #13: RPI Pico Carrier Board

epsi1on, 2024-03-26:
Hi,
Thanks for great project. could you suggest a simple carrier board for RPI pico to turn it into oscilloscope?
A very simple one like [THIS](https://time4ee.com/news.php?readmore=443) could work with PICO version?
I am also wrapping up to make next version of C# oscilloscope.

FilipDominec, 2024-03-26:
Sure, the analog input part (https://chiptron.cz/images/news/HS101/hs101_schematics.jpg) will protect your RP2 from being zapped by overvoltage, and will divide input by 6 (so full range of the ADC is again about 20 V). I don't know if this is the carrier board what you mean. 

If it were up to me, I would probably add a reasonably fast opamp to the input with say 10× amplification, positive/negative zero level shift, and also the input voltage divider should have a switch that would change voltage ranges like 2V, 5V, 10V, 20V, 50V... 

Just like on a real oscilloscope.

epsi1on, 2024-03-27:
i think by voltage divider with switch, you mean something like this
[image]
I was thinking as transistor is kind of switch, do you think that this divider could digitally be controlled with PWM pins and set of transistors instead of manual switches?
The PWM pins of Pico are free and i think i can use them for this purpose.

FilipDominec, 2024-04-06:
If you are OK with this, let's close the topic. 

When I make some reference design for a future rp2daq-oscilloscope, I will post it in a separate project.

Further electrical engineering may be better discussed by e-mail, or even better, on https://electronics.stackexchange.com/

closed 2024-04-06 by FilipDominec

epsi1on, 2024-08-13:
just for information, i did some dummy design for an elementary oscilloscope hardware for RPI pico without any active component! (used components are limited to resistor, capacitor, diode, switch and BNC connector. ([link](https://github.com/epsi1on/RpiPicoOsciloscope/tree/main/hardware))
I'm very happy with it, works very well with a good USB cable.
Thanks for sharing this project.

FilipDominec, 2024-08-13:
Good. I am still having the 100Msps oscilloscope/log-analyzer in mind, though there are much more scientific tasks to be done before.

epsi1on, 2024-08-13:
> Good. I am still having the 100Msps oscilloscope/log-analyzer in mind, though there are much more scientific tasks to be done before.
That one would be great.
I think the wpf UI have reasonable performance. it gets continues 500k samples without any missing data.
The scope did even visualized a 120 Mhz sin wave (produced by a SI5351 module)  on the monitor like a normal sin wave! (when i directly connected the signal to ADC pin without any resistor in between).
but due to Nyquist theorem it didn't detect the right frequency but the sin wave shape was almost perfectly detected...
I'll wait for the 100M scope, if it is USB based then I think can do some help...