FilipDominec/rp2daq #21: Sampling with two channels

AlessioCarnemolla, 2024-11-20:
I tried to sample two different signals with two channels (GPIO26, GPIO27) using example_ADC_async.py. One of them was linked to a Function Generator and the other was linked with a Jumper to GP2. Processing the two signals it came out they are the same signal and, doing the FFT, it shows they have same peak frequency. I didn't modify any line of the code. Is there something I have to add? How can i record two different signals precisely?

FilipDominec, 2024-11-20:
Hi, thanks for reporting. I will check it on my side. But first I must ask:

1. Did you use the stable release 0.2, or the latest github version of rp2daq? 
2. What script did you use, was it [example_ADC_async.py](https://github.com/FilipDominec/rp2daq/blob/main/example_ADC_async.py) or [example_ADC_sync_plot.py](https://github.com/FilipDominec/rp2daq/blob/main/example_ADC_sync_plot.py) or just the short snippet from the [README](https://github.com/FilipDominec/rp2daq?tab=readme-ov-file#receiving-analog-data)?
3. Are the results really 1-to-1 numerically the same (i.e. doubled values), or is there some little random difference (i.e. ADC properly sampling the same analog signal)?

AlessioCarnemolla, 2024-11-20:
Hi, I used the latest release of rp2daq. I used the 'example_ADC_async.py' code. There are some difference between the values of the signal but the shape is the same, the mean value is almost the same and the peak frequency is precisely the same. Here the graphs and the post processing done by Matlab. Thanks a lot for your avaliability.

[image]
[image]
[image]
[image]
[image]

AlessioCarnemolla, 2024-11-20:
Now the FG produces a wave of 17Hz Peak and 3V pk-pk on GPIO27. GPIO26 is still connected to GP2. This is what I get (peak frequency in the graphs corresponds to 17Hz).
[image]
[image]
[image]
[image]
[image]

AlessioCarnemolla, 2024-11-20:
#!/usr/bin/python3  
#-*- coding: utf-8 -*-
"""
This slightly fancier example receives multiple data blocks from the built in ADC 
(analog-to-digital converter) Raspberry Pi Pico.

It demonstrates how a virtually unlimited amount of data can be acquired, with
the main thread staying responsive. Such asynchronous, callback-based operation is 
generally more flexible and efficient. 

Rp2daq implements direct-memory access (DMA), triple-buffering and bit-compression 
in the firmware, along with multi-threading and -processing in the python module, to 
achieve reliable uninterrupted data stream at 500k × 12bit samples per second.

Finally, the ADC record is separated into channels and plotted using numpy and
matplotlib's interactive plot.

        Filip Dominec 2023, public domain
"""

## User options
ADC_channel_names = {0:"GPIO 26", 1:"GPIO 27", 2:"GPIO 28", 3:"ref V", 4:"builtin thermo"}

channels = [0,1]     # 0,1,2 are GPIOs 26-28;  3 is V_ref and 4 is internal thermometer
kSPS_per_ch = 10 * len(channels)    # note there is only one multiplexed ADC

import rp2daq 
import threading
import time

## Connect to the device
rp = rp2daq.Rp2daq()

## Generate some realistic signal on GPIO 2 (connect it with a jumper to GPIO 26)
rp.pwm_configure_pair(gpio=2, wrap_value=2**16-1, clkdiv=20, clkdiv_int_frac=0)
rp.pwm_set_value(gpio=2, value=3500) # minimum position
time.sleep(.1)

## Run-time objects and variables
all_ADC_done = threading.Event() # a thread-safe semaphore
all_data = []
received_time = []
delayed = []

# Called from other thread whenever data come from RP2
prev_etime_us = 0
def ADC_callback(**kwargs): 
    global t0, prev_etime_us

    all_data.extend(kwargs['data'])
    
    
    delayed.extend([kwargs['block_delayed_by_usb']]*(len(kwargs['data'])//2))

    print("Packet received", len(all_data), len(kwargs['data']),
            -kwargs['start_time_us']+kwargs['end_time_us'],
            -prev_etime_us+kwargs['start_time_us'],
            rp._i.report_queue.qsize(), 
            (" DELAYED" if kwargs['block_delayed_by_usb'] else "") )
    prev_etime_us = kwargs['end_time_us']
    received_time.extend([time.time()]*(len(kwargs['data'])//len(channels)))

    if not kwargs['blocks_to_send']: # i.e. if all blocks were received
        all_ADC_done.set()   # releases wait() in the main tread

    if not t0:
        t0 = time.time()

    #print(f"at {time.time()-t0:.3f} s, received {len(kwargs['data'])} new " +
            #f"ADC values ({kwargs['blocks_to_send']} blocks to go)" + 
            #(" DELAYED" if kwargs['block_delayed_by_usb'] else "") )

## Initialize the ADC into asynchronous operation...
t0 = None
rp.adc(channel_mask=sum(2**ch for ch in channels), 
        blocksize=1000*len(channels), 
        blocks_to_send=20, 
        #trigger_gpio=1,
        trigger_on_falling_edge=1,
        clkdiv=int(48000//kSPS_per_ch), 
        _callback=ADC_callback)
## Unless all data are received, the program can continue (or wait) here. A dedicated separate process
## ensures that raw data are quickly offloaded from USB into a queue, so that no message is corrupted. 
## High CPU load in this user script can however lead to delays in the callbacks being issued.
## In particular, tight busy loops in main thread (option 4 below) will cause callback delays, so 
## maximum sustained data rate may be roughly halved on ordinary modern computer (<300 kBps).

#all_ADC_done.wait() ## Waiting option 1: the right and efficient waiting (data rate OK, no loss)

#while not all_ADC_done.is_set(): # Waiting option 2: moderate CPU load is (also OK)
    #time.sleep(.000005)
    
def busy_wait(t): # Waiting option 3: stress test with busy loops (still OK)
    t0 = time.time()
    while time.time() < t0+t: pass
while not all_ADC_done.is_set():
    rp.gpio_out(25,1)
    busy_wait(.01)
    rp.gpio_out(25,0)
    busy_wait(.01)

#rp.gpio_out(25,1) # Waiting option 4: stress test with single busy loop
#while not all_ADC_done.is_set(): pass
#rp.gpio_out(25,0) # note this only happens after all data is received on computer side

print(f"Received total {len(all_data)} samples in {time.time()-t0}")
print(f"Average processed data rate was {len(all_data)/(time.time()-t0):.2f} samples per second")

## Optional plotting of all channels
t0 = time.time()
import matplotlib.pyplot as plt
import numpy as np

def adc_to_temperature(adc, Vref=3.30): 
    return 27 - (adc*Vref/4095 - 0.716)/0.001721

print(f"Plotting took {time.time() - t0:.3f} s")
y = np.array(all_data)
fig, ax = plt.subplots(nrows=1, ncols=1, figsize=(12, 10))
for ofs,ch in enumerate(channels):
    ax.plot(y[ofs::len(channels)] * 3.3 / 2**12, 
            label=ADC_channel_names[ch],
            lw=.3, c='rgbycm'[ch]) # marker=',', 
    if ch==4: print(f"Average temperature at channel no.4 = {adc_to_temperature(np.mean(y[ofs::len(channels)])):.3f} °C")

#ax.plot((-min(received_time)+np.array(received_time)), lw=.3, c='k', label='message received time')  # timing
ax.plot(np.array(delayed), lw=.8, c='k', label='(message delayed warning)') 

ax.set_xlabel("ADC count per channel")
ax.set_ylabel("ADC readout (V)")
ax.legend()
plt.grid(True)
plt.show()

import csv
with open('DataAsync.csv', 'w', newline='') as csvfile:
    writer = csv.writer(csvfile)
    for i in range(0, len(all_data), len(channels)):
        group = all_data[i:i+len(channels)]
        formatted_group = [f"{(x * 3.3 / (2 ** 12)):.8f}" for x in group]
        writer.writerow(formatted_group)

print('File updated')

this is the code I used. I've only added the last ten lines to save the data on a .csv file.

AlessioCarnemolla, 2024-11-21:
It seems that the code detects only one frequency from the two channel, printing one of them for both, casually. May it be a hardware problem? I tried with different jumpers.

FilipDominec, 2024-11-26:
I am very busy now, but I will try to replicate & fix it until Friday. Thanks for understanding.

AlessioCarnemolla, 2024-11-26:
Thank you a lot, for your avaliability.

FilipDominec, 2024-11-29:
Since you are using the previous release 0.2.0, I downgraded the firmware and run a similar script from the release's directory. And I got basically the same result. I guess there might be some hardware trouble with your device? 

[image]

FilipDominec, 2025-02-28:
@AlessioCarnemolla I can confirm the signal on both channels comes from the first user-specified channel. The channels are slightly shifted, so it's coming from properly sampled values being wrongly interpreted as two interlaced channels. 

I will try to fix this over the weekend. Then I will make a new 0.3.0 release, if there are no other troubles.

FilipDominec, 2025-03-11:
@AlessioCarnemolla I will be releasing the new firmware soon and it seems fixed; will you be available to test it in your setup?

AlessioCarnemolla, 2025-03-12:
Yes, I will try to sample with two microphone capsules. Thank You

AlessioCarnemolla, 2025-03-17:
I tried this morning and after download the last firmware version it printed this " rp2daq device firmware has version 231005,
older than this script requires: 250215". 

I tried many times to download the latest version from the link in the README file but it seems the latest version is the 231005 version.

FilipDominec, 2025-03-17:
This looks like an unclear installation guide in the README. I will rewrite it to make it clear.

The downloaded Python+C source code and the uploaded firmware versions must match. The matching versions of both are available at
https://github.com/FilipDominec/rp2daq/releases/tag/v0.2.0
and are tested to be matching.

I will finally publish a 0.3.0 release today, so you will be able to try out the new features. The API will change a bit, though, so you may want to adapt your scripts. Or keep using the 0.2.0 if it is fine for you.

F.

FilipDominec, 2025-03-17:
@AlessioCarnemolla, if you have a minute, please try the 0.3.0 release. All issues with internal ADC should be fixed!